ADLab-AutoDrive/BEVFusion #2: Camera failure test

alexa19lab, 2022-06-06:
@ADLab-AutoDrive  Thanks so much for the simple and effective work and releasing the code ! After reading your paper, I am curious about the camera failure experiments. May I ask if it is possible for me to reproduce the camera failure experiments with the code and model that you currently provide? Many thanks!!

kcyu2014, 2022-06-06:
Thanks for your interest! For the robustness scenarios, we compose a formal robustness benchmark for lidar-camera fusion models. Please follow this [repo](https://github.com/ADLab-AutoDrive/lidar-camera-robust-benchmark) and paper at https://arxiv.org/abs/2205.14951 for more details. We will release the full benchmark by the end of June 2022.

alexa19lab, 2022-06-06:
> Thanks for your interest! For the robustness scenarios, we compose a formal robustness benchmark for lidar-camera fusion models. Please follow this [...]

Thanks for such useful tools and benchmark! May I ask whether the robustness evaluation is based on one model or it is based on different models trained with datasets with different noises? In other word, do we need to keep the consistency between training and evaluation dataset?

tingtingliangvs, 2022-06-09:
In this paper, data augmentation is consistent between training and evaluation datasets. You are welcome to train and evaluate models under the combined augmentation strategies and share your findings with us.

closed 2022-12-28 by tingtingliangvs