ADLab-AutoDrive/BEVFusion #4: Training Error

Treemann, 2022-06-09:
Hi,
Nice work!

I could run the inference code successfully, but encountered with errors during training.
Issue1: 
```
    raise ModuleAttributeError("'{}' object has no attribute '{}'".format(
torch.nn.modules.module.ModuleAttributeError: 'BEVF_FasterRCNN' object has no attribute 'kd'
```
I found `self.kd` and `self.kd_feat_loss` not defined, so I added `self.kd=False` in the __init__() function. 
```
            if self.kd:
                losses_pts['kd_feat_loss'] = self.kd_feat_loss
```
Issue2: 
```
    Variable._execution_engine.run_backward(
RuntimeError: CUDA out of memory. Tried to allocate 3.94 GiB (GPU 4; 31.75 GiB total capacity; 17.30 GiB already allocated; 3.86 GiB free; 26.34 GiB reserved in total by PyTorch)
```
I used 8xV100(32G) and the error above occurred, I tried with adding `torch.cuda.empty_cache()` to the training script, but still OOM. Then I tried to tune the parameter `sample_per_gpu: 4 → 2` for training.

tingtingliangvs, 2022-06-09:
Thank you for reporting the bug. It is because I did some ablation study and it didn't work, but forgot to fully erase it. We have fixed it.
Would you please describe the specific config which OOM during training?

Treemann, 2022-06-09:
I use this config to train the fusion model: [bevf_pp_4x8_1x_nusc.py](https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/configs/bevfusion/bevf_pp_4x8_1x_nusc.py)

Treemann, 2022-06-09:
Got it! Thanks for your detailed explanation.

Treemann, 2022-06-10:
Hi @tingtingliangvs ,
I train the fusion model according to your config [bevf_pp_2x8_1x_nusc.py](https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/configs/bevfusion/bevf_pp_2x8_1x_nusc.py), loading the released 2D/3D pretrained models.
And there is a gap between my result with the claimed one, is this the config for the claimed accuracy?
|Model|mAP|NDS|
|-|-|-|
|claimed|53.5 | 60.4|
|reproduced|50.3|56.8|

Treemann, 2022-06-14:
Hi @tingtingliangvs , is the gap reasonable for the config discussed ?

tingtingliangvs, 2022-06-15:
Hi, I check my training log, and the learning rate for pointpillars should be 0.001.  I retrain [bevf_pp_2x8_1x_nusc.py](https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/configs/bevfusion/bevf_pp_2x8_1x_nusc.py) get the reproduced result at epoch 9.
Model | mAP | NDS
-- | -- | --
claimed | 53.5 | 60.4
reproduced | 54.0 | 60.5

Hope it helps.

Treemann, 2022-06-16:
Got similar result (mAP: 53.7, NDS: 60.7) with the claimed accuracy by tuning the lr.
Thanks for your reply~

closed 2022-06-16 by Treemann

Birdylx, 2022-11-14:
@Treemann hi, I'm trying reproduce the lidar stream results, I used the command which authors provide, but get lower mAP and NDS, can you reproduce the lidar stream results?