ADLab-AutoDrive/BEVFusion #5: about camera calibration

MasatoSaeki, 2022-06-18:
Thank you for your working.

1.
https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/mmdet3d/models/detectors/bevf_faster_rcnn.py#L126
2.
https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/mmdet3d/models/detectors/cam_stream_lss.py#L311

In 1., the projection matrix is passed, but in 2., it looks like it is not used.
Do you use this kind of camera calibration when projecting to BEV?
(In original liftsplatshoot, that is used)

tingtingliangvs, 2022-07-11:
Hi, we use camera extrinsic parameters 'rots' and 'trans' when projecting to BEV. Since no data augmentation is used, we do not need to recover 'post_trans' and 'post_rots'.

jcuic5, 2022-07-11:
Hi, I also have a question. Since 'rots' and 'trans' are extrinsic transformations, doesn't it still needs the intrinsics to perform the projection? Seems the pre-generated frustum grid is with pixel coordinate values

MasatoSaeki, 2022-07-16:
Thank you very much.

closed 2022-07-16 by MasatoSaeki

hrz2000, 2022-09-30:
> Hi, we use camera extrinsic parameters 'rots' and 'trans' when projecting to BEV. Since no data augmentation is used, we do not need to recover 'post_trans' and 'post_rots'.

In LSS's code：
[image]

but your code don‘t have the ‘intrins’，why？

zhanghaozhe23, 2023-03-02:
Have you solved the problem?