ADLab-AutoDrive/BEVFusion #8: how to demo or inference

ChunmianLin, 2022-07-11:
Hi, how to perform a demo or model inference to check the bev map generation process in the image stream? Also, the data_preparation seems to be missed？ Could you please provide a new guidance for dataset file structure? Thanks.

tingtingliangvs, 2022-07-11:
Hi, 
`./tools/dist_test.sh configs/bevfusion/cam_stream/bevf_pp_4x8_2x_nusc_cam.py ./work_dirs/cam_pp.pth 8 --eval bbox` performs camera stream inference. 
We follow the data preparation in [mmdet3d](https://github.com/open-mmlab/mmdetection3d/blob/master/docs/en/datasets/nuscenes_det.md), please follow the file structure.
Please refer to [mmdet3d](https://github.com/open-mmlab/mmdetection3d/blob/master/docs/en/getting_started.md) to get started.

tingtingliangvs, 2022-07-12:
Hi, you may follow the newest [inference demo](https://github.com/open-mmlab/mmdetection3d/blob/master/demo/inference_demo.ipynb) in mmdet3d to change the data input in [tools/test.py](https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/tools/test.py).

ChunmianLin, 2022-07-12:
many thanks! I have performed model inference on mini dataset.

closed 2022-12-28 by tingtingliangvs