ADLab-AutoDrive/BEVFusion #9: Question cocerning the pretrained model.

Solacex, 2022-07-12:
Hello,  thanks for your inspiring work~

For training BEVfusion with the config configs/bevfusion/bevf_pp_2x8_1x_nusc.py, we notice the pretrained model of camera stream is missing, I,e., work_dirs/bevf_pp_4x8_2x_nusc_cam/epoch_24.pth. 

Have you provided this model in this repo? Or we can use the provided mask_rcnn_dbswin-t_fpn_3x_nuim_cocopre.pth directly?

tingtingliangvs, 2022-07-12:
Hi, the pretrained camera stream is in the [readme](https://github.com/ADLab-AutoDrive/BEVFusion#:~:text=31.1-,Model,-BEVFusion), we have uploaded the weights in [google drive](https://drive.google.com/file/d/1OeDjGoKpU1FQ42_18pJZ-UqwxLDc6HRc/view).

Solacex, 2022-07-12:
So it should be cam_pp.pth.

Thanks for your prompt reply!

closed 2022-07-12 by Solacex