ADLab-AutoDrive/BEVFusion #10: How about freezing the LiDAR stream?

Solacex, 2022-07-13:
Hello,

I find a trigger freeze_lidar_components in the config file. Have you performed the BEVfusion [the second stage] with the LiDAR stream frozen? Or freezing both LIDAR and camera stream and optimizing the fusion part only?

tingtingliangvs, 2022-12-28:
Hi, we freeze both LIDAR and camera stream and optimize the fusion part only.

HaoZhang16613, 2023-08-03:
I have some questions about the training strategy. Why did you choose to freeze other parts and only optimize the fusion module? Is it because the model may suffer from overfitting? 
Additionally, why freeze the head part? In my understanding, the head part should also be optimized on the fused features.
Looking forward to your response. Thank you very much!