ADLab-AutoDrive/BEVFusion #18: How much can the img_depth improve the NDS?

chdzhen, 2022-09-09:
How much can the img_depth improve the NDS?

in bevf_transfusion.py
loss_depth = self.depth_dist_loss(depth_dist, img_depth, loss_method=self.img_depth_loss_method, img=img) * self.img_depth_loss_weight
losses.update(img_depth_loss=loss_depth)

thanks!

tingtingliangvs, 2022-12-28:
Hi, in our previous experiment, the img_depth shows no improvement to the NDS. For more implementation results, please refer to [BEVDepth](https://github.com/Megvii-BaseDetection/BEVDepth) and other works.

HaoZhang16613, 2023-04-11:
> Hi, in our previous experiment, the img_depth shows no improvement to the NDS. For more implementation results, please refer to [BEVDepth](https://github.com/Megvii-BaseDetection/BEVDepth) and other works.

Hi~ when I was training the camera branch (bevf_pp_4x8_2x_nusc_cam) with img_depth, loss became nan after serveral iters.
I've already tried some solutions, like adding img_depth loss after a few training epoches and setting img_depth loss to zero when it is abnormal ... Unfortunately, all solutions didn't work.