ADLab-AutoDrive/BEVFusion #19: About Transfusion pretrained model?

serend1p1ty, 2022-10-10:
Hello, great work! Can you provide the Transfusion pretrained model?

tingtingliangvs, 2022-11-02:
Hi, I upload the LiDAR, Camera, Fusion model of TransFusion in [google drive](https://drive.google.com/file/d/1QfUV7cLfNMdumOX4xxNjaaS3LoxYxjH-/view?usp=sharing), hope it helps.

serend1p1ty, 2022-11-06:
Thank you very much!

closed 2022-11-06 by serend1p1ty

zlenyk, 2023-10-11:
Hi! Just wanted to clarify what was the LR used for cam-only training.
There are 2 posted configs: 
- https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/configs/bevfusion/cam_stream/bevf_tf_4x8_20e_nusc_cam.py
- https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/configs/bevfusion/cam_stream/bevf_tf_4x8_20e_nusc_cam_lr.py

In the posted models starting LR seems to be 1e-4, but in config posted in "Benchmark Evaluation and Training" it is 1e-5.
Could you clarify?

Thank you!

Libraaer, 2024-08-31:
> Hi, I upload the LiDAR, Camera, Fusion model of TransFusion in [google drive](https://drive.google.com/file/d/1QfUV7cLfNMdumOX4xxNjaaS3LoxYxjH-/view?usp=sharing), hope it helps.

你好，请问下载的三个文件中 lidar_tf.pth、 cam_tf.pth、 bevfusion_tf.pth具体是指什么呢？ 例如lidar_tf.pth是指configs/bevfusion/bevf_tf_4x8_6e_nusc.py 中的work_dirs/transfusion_nusc_voxel_L/epoch_20.pth这个文件吗？还有bevfusion_tf.pth是指什么文件呢？您这里是否有work_dirs/bevf_tf_4x8_20e_nusc_cam.py/epoch_20.pth这个文件呢？如果有得话请给我发一份，谢谢您。