ADLab-AutoDrive/BEVFusion #20: import 'ball_query_ext' issue

buaazeus, 2022-10-10:
I tried to train the model and got this issue.
cannot import name 'ball_query_ext' from partially initialized module 'mmdet3d.ops.ball_query'
Anyone has got and fixed this issue?
Thank you!

hongweiliuideal, 2022-10-13:
hello，the environment is not compiled successfully, it is recommended to use **python setup.py develop** to compile successfully and then run the code

buaazeus, 2022-10-24:
Issue is fixed. Thank you.

closed 2022-10-24 by buaazeus

zjufkq, 2022-10-25:
I also have this problem But it is not work

zjufkq, 2022-10-25:
> hello，the environment is not compiled successfully, it is recommended to use **python setup.py develop** to compile successfully and then run the code

mmdet3d?

buaazeus, 2022-10-25:
python setup.py develop in BEVFusion path.