ADLab-AutoDrive/BEVFusion #23: missing keys in source state_dict

BHC1205, 2022-10-14:
Hi,I came across the following warning in Training,please take a look about it,Thanks
The camera-only model : https://github.com/ADLab-AutoDrive/BEVFusion#nuscenes-detection-validation
print(cfg.load_lift_from) is ok

missing keys in source state_dict: img_backbone.cb_modules.0.patch_embed.proj.weight, img_backbone.cb_modules.0.patch_embed.proj.bias, img_backbone.cb_modules.0.patch_embed.norm.weight, img_backbone.cb_modules.0.patch_embed.norm.bias, img_backbone.cb_modules.0.layers.0.blocks.0.norm1.weight, img_backbone.cb_modules.0.layers.0.blocks.0.norm1.bias, img_backbone.cb_modules.0.layers.0.blocks.0.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.0.blocks.0.attn.relative_position_index, img_backbone.cb_modules.0.layers.0.blocks.0.attn.qkv.weight, img_backbone.cb_modules.0.layers.0.blocks.0.attn.qkv.bias, img_backbone.cb_modules.0.layers.0.blocks.0.attn.proj.weight, img_backbone.cb_modules.0.layers.0.blocks.0.attn.proj.bias, img_backbone.cb_modules.0.layers.0.blocks.0.norm2.weight, img_backbone.cb_modules.0.layers.0.blocks.0.norm2.bias, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc1.weight, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc1.bias, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc2.weight, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc2.bias, img_backbone.cb_modules.0.layers.0.blocks.1.norm1.weight, img_backbone.cb_modules.0.layers.0.blocks.1.norm1.bias, img_backbone.cb_modules.0.layers.0.blocks.1.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.0.blocks.1.attn.relative_position_index, 
....
lift_splat_shot_vis.dx, lift_splat_shot_vis.bx, lift_splat_shot_vis.nx, lift_splat_shot_vis.frustum, lift_splat_shot_vis.camencode.depthnet.weight, lift_splat_shot_vis.camencode.depthnet.bias, lift_splat_shot_vis.bevencode.0.weight, lift_splat_shot_vis.bevencode.1.weight, lift_splat_shot_vis.bevencode.1.bias, lift_splat_shot_vis.bevencode.1.running_mean, lift_splat_shot_vis.bevencode.1.running_var, lift_splat_shot_vis.bevencode.3.weight, lift_splat_shot_vis.bevencode.4.weight, lift_splat_shot_vis.bevencode.4.bias, lift_splat_shot_vis.bevencode.4.running_mean, lift_splat_shot_vis.bevencode.4.running_var, lift_splat_shot_vis.bevencode.6.weight, lift_splat_shot_vis.bevencode.7.weight, lift_splat_shot_vis.bevencode.7.bias, lift_splat_shot_vis.bevencode.7.running_mean, lift_splat_shot_vis.bevencode.7.running_var, lift_splat_shot_vis.bevencode.9.weight, lift_splat_shot_vis.bevencode.10.weight, lift_splat_shot_vis.bevencode.10.bias, lift_splat_shot_vis.bevencode.10.running_mean, lift_splat_shot_vis.bevencode.10.running_var, seblock.att.1.weight, seblock.att.1.bias, reduc_conv.conv.weight, reduc_conv.bn.weight, reduc_conv.bn.bias, reduc_conv.bn.running_mean, reduc_conv.bn.running_var

tingtingliangvs, 2022-11-02:
Hi, could you describe which config file you are using?

zjufkq, 2023-03-06:
Hi,Have you solved it? I have the same problem.
I use the weight parameter provided by the author cam_pp.pth and hv_ pointpilars...pth(github) ，then runs the fusion stream code (run this command)
./tools/dist_ train.sh configs/bevfusion/bevf_ pp_ 2x8_ 1x_ nusc.py 8
The above warning will appear:

mmdet - WARNING - The model and loaded state dict do not match exactly
missing keys in source state_dict: img_backbone.cb_modules.0.patch_embed.proj.weight, img_backbone.cb_modules.0.patch_embed.proj.bias, img_backbone.cb_modules.0.patch_embed.norm.weight, img_backbone.cb_modules.0.patch_embed.norm.bias, img_backbone.cb_modules.0.layers.0.blocks.0.norm1.weight, img_backbone.cb_modules.0.layers.0.blocks.0.norm1.bias, img_backbone.cb_modules.0.layers.0.blocks.0.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.0.blocks.0.attn.relative_position_index, img_backbone.cb_modules.0.layers.0.blocks.0.attn.qkv.weight, img_backbone.cb_modules.0.layers.0.blocks.0.attn.qkv.bias, img_backbone.cb_modules.0.layers.0.blocks.0.attn.proj.weight, img_backbone.cb_modules.0.layers.0.blocks.0.attn.proj.bias, img_backbone.cb_modules.0.layers.0.blocks.0.norm2.weight, img_backbone.cb_modules.0.layers.0.blocks.0.norm2.bias, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc1.weight, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc1.bias, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc2.weight, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc2.bias, img_backbone.cb_modules.0.layers.0.blocks.1.norm1.weight, img_backbone.cb_modules.0.layers.0.blocks.1.norm1.bias, img_backbone.cb_modules.0.layers.0.blocks.1.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.0.blocks.1.attn.relative_position_index, img_backbone.cb_modules.0.layers.0.blocks.1.attn.qkv.weight, img_backbone.cb_modules.0.layers.0.blocks.1.attn.qkv.bias, img_backbone.cb_modules.0.layers.0.blocks.1.attn.proj.weight, img_backbone.cb_modules.0.layers.0.blocks.1.attn.proj.bias, img_backbone.cb_modules.0.layers.0.blocks.1.norm2.weight, img_backbone.cb_modules.0.layers.0.blocks.1.norm2.bias, img_backbone.cb_modules.0.layers.0.blocks.1.mlp.fc1.weight, img_backbone.cb_modules.0.layers.0.blocks.1.mlp.fc1.bias, img_backbone.cb_modules.0.layers.0.blocks.1.mlp.fc2.weight, img_backbone.cb_modules.0.layers.0.blocks.1.mlp.fc2.bias, img_backbone.cb_modules.0.layers.0.downsample.reduction.weight, img_backbone.cb_modules.0.layers.0.downsample.norm.weight, img_backbone.cb_modules.0.lay

>