ADLab-AutoDrive/BEVFusion #24: KeyError: 'cam_intrinsic'

hongweiliuideal, 2022-10-18:
HI，Great Work！
When I was training, I suddenly reported an error in the middle of the operation. I encountered it in the training camera flow, lidar flow and bevfusion. How to solve it.

The following is the error：
KeyError: 'cam_intrinsic'

hongweiliuideal, 2022-10-26:
run create_data.py

hongweiliuideal, 2022-10-26:
yes

Birdylx, 2022-11-12:
Hi, @idealliu, can you reproduce the lidar stream results? I just run the command 
`./tools/dist_train.sh configs/bevfusion/lidar_stream/hv_pointpillars_secfpn_sbn-all_4x8_2x_nus-3d.py 8` to train lidar stream. But I get lower mAP and NDS, I wonder that can you reproduce the lidar stream results?

idealliu, 2022-12-24:
> Hi, @idealliu, can you reproduce the lidar stream results? I just run the command `./tools/dist_train.sh configs/bevfusion/lidar_stream/hv_pointpillars_secfpn_sbn-all_4x8_2x_nus-3d.py 8` to train lidar stream. But I get lower mAP and NDS, I wonder that can you reproduce the lidar stream results?

hi，can you solve this problem：keyerror：‘cam_intrinsic’, I have run create.py,but  this problem didn't solved,How did you solve it

hwhongweiliu, 2022-12-24:
> Hi, @idealliu, can you reproduce the lidar stream results? I just run the command `./tools/dist_train.sh configs/bevfusion/lidar_stream/hv_pointpillars_secfpn_sbn-all_4x8_2x_nus-3d.py 8` to train lidar stream. But I get lower mAP and NDS, I wonder that can you reproduce the lidar stream results?

hi，can you solve this problem：keyerror：‘cam_intrinsic’, I have run create.py,but this problem didn't solved,How did you solve it