ADLab-AutoDrive/BEVFusion #26: pretrain model missing

buaazeus, 2022-10-26:
work_dirs/mask_rcnn_cbv2_swin_tiny_patch4_window7_mstrain_480-800_adamw_3x_coco/mask_rcnn_cbv2_swin_tiny_patch4_window7_mstrain_480-800_adamw_3x_coco.pth is not a checkpoint file

Hello, I got and pretrain model missing issue.
Anyone know where to find this model file?
mask_rcnn_cbv2_swin_tiny_patch4_window7_mstrain_480-800_adamw_3x_coco.pth
Thank you.

buaazeus, 2022-10-26:
your link is  mask_rcnn_dbswin-t_fpn_3x_nuim_cocopre.pth. seems not same file...

tingtingliangvs, 2022-11-02:
Hi, they are the same file with different names. dbswin-t is cbv2_swin_tiny.

buaazeus, 2022-11-02:
Got it, Thank you.

closed 2022-11-02 by buaazeus

jamesgunnfiveai, 2023-01-20:
Hi @tingtingliangvs - where does the mask_rcnn_dbswin-t_fpn_3x_nuim_cocopre.pth checkpoint come from? Is it something you pretrained? If so do you have a link to how this was done? Or has it come from another project?

jamesgunnfiveai, 2023-01-22:
Ah, my apologies, I can see now that it was trained using https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/configs/bevfusion/cam_stream/mask_rcnn_dbswin-t_fpn_3x_nuim_cocopre.py and starting from https://github.com/SwinTransformer/storage/releases/download/v1.0.0/swin_tiny_patch4_window7_224.pth