ADLab-AutoDrive/BEVFusion #27: mmdet3d Version Incompatible

zjufkq, 2022-10-26:
I can't install mmdet3d=0.11.0 with the recommended version of torch=1.7.0 cuda=10.1 or 11.0. What's the matter? Or has anyone successfully configured the environment? thank
If I use torch=1.6.0, I can install mmdet3d==0.11.0, but code errors will occur in the end

zjufkq, 2022-10-27:
I GOT IT

closed 2022-10-27 by zjufkq

mqm-slice, 2022-11-01:
So what version configuration did you use？Can you share with me?

eyabesbes, 2024-04-13:
> I can't install mmdet3d=0.11.0 with the recommended version of torch=1.7.0 cuda=10.1 or 11.0. What's the matter? Or has anyone successfully configured the environment? thank If I use torch=1.6.0, I can install mmdet3d==0.11.0, but code errors will occur in the end

Hi,
I'm also facing the same problem?
how did you manage to solve the issue ?

eyabesbes, 2024-04-13:
> So what version configuration did you use？Can you share with me?

Hi,
did you manage to configure the environment ?