ADLab-AutoDrive/BEVFusion #32: Differences in inference results

Young98CN, 2022-11-05:
This is a great job, but I have two questions 

1. There is a bug in running the [configuration file](https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/configs/bevfusion/bevf_tf_4x8_10e_nusc_aug.py).
2. Why I downloaded the pre-trained model that you [provided](https://drive.google.com/file/d/1wgcR6oYCrZvo3-whA5S4obTHce7aJ7OU/view?usp=share_link) and the inference results did not match the readme **NDS：72.1 mAP：69.6** . I got the result of **NDS：0.7132 mAP：0.6848** with [config](https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/configs/bevfusion/bevf_tf_4x8_6e_nusc.py)（Maybe I didn't find the difference between this configuration file and the above mentioned）

Looking forward to your reply ^_^

Young98CN, 2022-11-06:
> This is a great job, but I have two questions
>
> 1. There is a bug in running the [configuration [...]

The link to the bug is [here](https://github.com/ADLab-AutoDrive/BEVFusion/blob/c1cf4f2b018d89f289164d07f57a42d8bcfbfba3/mmdet3d/models/detectors/bevf_faster_rcnn_aug.py#L177)
`TypeError: forward() got an unexpected keyword argument 'extra_rots'`

Trent-tangtao, 2022-11-07:
Thank you for your interest!  It seems that the bugs are produced by reorganization of the code.
Besides, I forgot to update the code of camera branch and upload the wrong checkpoint. 
These problems have been fixed. You can pull the commits and  reload the checkpoint,  then rerun it now.

Young98CN, 2022-11-07:
Thank you for your reply, but I noticed that the last update was five days ago, has the update been completed?

Trent-tangtao, 2022-11-07:
Please checkout the dev_aug branch ~

Young98CN, 2022-11-07:
Thanks for your patience, I will try it later! ^_^

Young98CN, 2022-11-07:
Maybe there is a problem with my implementation, I got the result of NDS：0.7132 mAP：0.6848 with [config](https://github.com/ADLab-AutoDrive/BEVFusion/blob/main/configs/bevfusion/bevf_tf_4x8_6e_nusc.py)(I load this [checkpoint](https://drive.google.com/file/d/1wgcR6oYCrZvo3-whA5S4obTHce7aJ7OU/view?usp=share_link) )

Trent-tangtao, 2022-11-07:
Hi, I have updated the checkpoint. Do you download the new model checkpoint?

Young98CN, 2022-11-07:
> Hi, I have updated the checkpoint. Do you download the new model checkpoint?

Oh, I downloaded the checkpoint under the main branch, Thanks for the reminder!

Young98CN, 2022-11-07:
Thank you for your patient answer, I have got the accuracy in the readme, and I will close this issue.

closed 2022-11-07 by Young98CN