ADLab-AutoDrive/BEVFusion #33: BEV space augmentation

kentang-mit, 2022-11-06:
Dear authors,

I'm one of the contributors to another project named BEVFusion. I noticed that you added in BEV space augmentation following BEVDet and our work. Thanks for citing our work in your up-to-date arXiv paper. 

I wonder if it is possible to also reflect that you have referred to our code when adding this feature to your codebase (discussion [here](https://github.com/mit-han-lab/bevfusion/issues/52) and [here](https://github.com/mit-han-lab/bevfusion/issues/22) back in June and July this year, and it seems that you did follow the same implementation in our codebase with the same variable names) in your README.md, thank you very much!

Best,
Haotian

Trent-tangtao, 2022-11-07:
Hi Haotian,

Thanks for your great efforts and the discussions with me.
As mentioned in our camera-ready version, we conduct BEV-space data augmentation following [[BEVDet](https://arxiv.org/abs/2112.11790)](https://arxiv.org/abs/2112.11790) and [[BEVFusion-mit](https://github.com/mit-han-lab/bevfusion)](https://github.com/mit-han-lab/bevfusion)(yours). And the arXiv paper will be updated later.
Sorry for my negligence to add the acknowledging of referenced implementations in our README.md of the codebase and I have updated it just now.
Thanks again for your attractive works and the great open-sourced code base.

Best,
Tangtao

kentang-mit, 2022-11-07:
Hi Tao,

Thank you very much! It's very helpful, and looking forward to your new arXiv version. I'll close this issue now.

Best,
Haotian

kentang-mit, 2022-11-10:
Dear authors,

When should I expect your new arXiv version? If you haven't finished, it would be greatly appreciated if you can discuss with us (Zhijian and me) about the paper update via Email next week such that we can avoid unnecessary back and forth. We'd like to see Prof. Wang and Prof. Tang involved in the discussion. Thanks for your time!

Best,
Haotian

tingtingliangvs, 2022-11-10:
Hi Haotian,

I have updated the arXiv and you will see the changes soon.
In the camera-ready version we uploaded to the NeurIPS system,  we didn't use the '70.2 mAP 72.9 NDS' MIT-Base version. Sorry for the trouble.

Best,
Tingting

kentang-mit, 2022-11-10:
Thank you @tingtingliangvs. We'll check the new content in the updated arXiv version and see if further discussions are needed. Have a nice day!

closed 2022-11-10 by kentang-mit