ADLab-AutoDrive/BEVFusion #35: Reproduce problem

Birdylx, 2022-11-08:
Hi, @tingtingliangvs @Trent-tangtao Thanks for your great work. I'm trying to reproduce the fusion results as mentioned in README Table. I follow the training strategy and use the command you provide, I get the same results in `camera stream`, but lower mAP, NDS in `lidar stream`, and this causes the fusion result much lower:
```
# I use pointpillar as 3D backbone and pointpillar head as detection head.
                            
bevf_pp_nusc_cam(repo):  mAP: 22.9. NDS: 31.1
bevf_pp_nusc_cam(reproduce):  mAP: 22.6. NDS: 30.6
bevf_pp_nusc_lidar(repo):  mAP: 35.1. NDS: 49.8
bevf_pp_nusc_lidar(reproduce):  mAP: 31.6. NDS: 47.6
bevf_pp_nusce_fusion(repo):  mAP: 53.5  NDS: 60.4
bevf_pp_nusce_fusion(reproduce):  mAP: 51.7. NDS: 59.5
```
Is there something that i missed? Do you have any advice to reproduce the results?

Best,
Birdy

Birdylx, 2022-11-12:
@tingtingliangvs @Trent-tangtao , can your provide config file `hv_pointpillars_secfpn_sbn-all_4x8_2x_nus-3d.py` or some log file in your output folder? I'm struggling doing lidar stream reproduce, I would appreciate it if you can provide some hints to produce the results.

Birdylx, 2022-11-24:
problem solved, you forgot to uncomment the data augmentation in lidar stream config,

closed 2022-11-24 by Birdylx

serend1p1ty, 2022-11-28:
@Birdylx Same issue here. Do you mean these lines?
[image]

JudasDie, 2022-11-30:
> problem solved, you forgot to uncomment the data augmentation in lidar stream config,

Hi, could you kindly provide the right config? met the same problem.

Birdylx, 2023-01-05:
@JudasDie @serend1p1ty ，that's right，you should uncomment these lines to open data augmentation