ADLab-AutoDrive/BEVFusion #41: cannot import name 'ball_query_ext' from partially initialized module 'mmdet3d.ops.ball_query' (most likely due to a circular import)

wistful-8029, 2022-11-18:
Hello, this problem occurred when I was organizing the dataset file. I read the source code and found no exceptions. No one asked in other issues. Is there a solution?

Traceback (most recent call last):
  File "/home/wistful/work/my_bevfusion/tools/create_data.py", line 5, in <module>
    from tools.data_converter import kitti_converter as kitti
  File "/home/wistful/work/my_bevfusion/tools/data_converter/kitti_converter.py", line 5, in <module>
    from mmdet3d.core.bbox import box_np_ops
  File "/home/wistful/work/my_bevfusion/mmdet3d/core/__init__.py", line 2, in <module>
    from .bbox import *  # noqa: F401, F403
  File "/home/wistful/work/my_bevfusion/mmdet3d/core/bbox/__init__.py", line 4, in <module>
    from .iou_calculators import (AxisAlignedBboxOverlaps3D, BboxOverlaps3D,
  File "/home/wistful/work/my_bevfusion/mmdet3d/core/bbox/iou_calculators/__init__.py", line 1, in <module>
    from .iou3d_calculator import (AxisAlignedBboxOverlaps3D, BboxOverlaps3D,
  File "/home/wistful/work/my_bevfusion/mmdet3d/core/bbox/iou_calculators/iou3d_calculator.py", line 5, in <module>
    from ..structures import get_box_type
  File "/home/wistful/work/my_bevfusion/mmdet3d/core/bbox/structures/__init__.py", line 1, in <module>
    from .base_box3d import BaseInstance3DBoxes
  File "/home/wistful/work/my_bevfusion/mmdet3d/core/bbox/structures/base_box3d.py", line 5, in <module>
    from mmdet3d.ops.iou3d import iou3d_cuda
  File "/home/wistful/work/my_bevfusion/mmdet3d/ops/__init__.py", line 5, in <module>
    from .ball_query import ball_query
  File "/home/wistful/work/my_bevfusion/mmdet3d/ops/ball_query/__init__.py", line 1, in <module>
    from .ball_query import ball_query
  File "/home/wistful/work/my_bevfusion/mmdet3d/ops/ball_query/ball_query.py", line 4, in <module>
    from . import ball_query_ext
ImportError: cannot import name 'ball_query_ext' from partially initialized module 'mmdet3d.ops.ball_query' (most likely due to a circular import)

closed 2022-11-18 by wistful-8029

Sadwy, 2023-02-06:
Hello. I met the same problem. May you share the solutions?

wistful-8029, 2023-02-12:
Sorry, I saw it too late. I haven't seen this project for a long time. I looked at my modified code and found that I did this. I can't remember anything else.

 

Wistfully
***@***.***