ADLab-AutoDrive/BEVFusion #42: CUDA out of memory

gubei1998, 2022-11-24:
When I train by bevf_tf_4x8_20e_nusc_cam_lr.py on 4x3090,samples_per_gpu=2, workers_per_gpu=4,it will report an error :CUDA out of memory,I would like to know what is causing this error, is it really due to a lack of video memory on the card or is it a code issue? I am looking forward to your answers,thank you very much!

foolhard, 2022-12-05:
Same as you. 
I tried all the camera stream training scripts in A100(40G), but all have  **out of CUDA memory** issue. 
I reduce the batch_size of `bevf_pp_4x8_2x_nusc_cam.py` to 2 per GPU and it works. The GPU memory cost is 31GB.

**How can I train `bevf_pp_4x8_2x_nusc_cam.py`  with 4 samples per GPU in V100(32G)?**

tingtingliangvs, 2022-12-28:
Hi, reducing the batch_size of bevf_pp_4x8_2x_nusc_cam.py to 2 per GPU with "freeze_img=False" should give better results than the original paper.
In previous experiments, bevf_pp_4x8_2x_nusc_cam.py with 4 samples per GPU worked in V100 (32G) with "freeze_img=True", but the accuracy was not satisfactory.
Thanks for the reminder, I'll fix this in the repo.

Birdylx, 2023-01-10:
@tingtingliangvs Hi, if `set freeze_img=False`, how much improvement it will get, if not freeze the cam_stream, it will cost more time to train right? how much training time it will be ?