ADLab-AutoDrive/BEVFusion #56: Error for VFE layer setting/configuration of Pointpillar model

FrontierBreaker, 2023-02-16:
https://github.com/ADLab-AutoDrive/BEVFusion/blob/3f992837ad659f050df38d7b0978372425be16ff/mmdet3d/models/voxel_encoders/voxel_encoder.py#L357

Hi, I met a bug of mismatching vfe layer parameter dimension with feature embedding dimension and I find the error is related to this line. 
As the outputted channel of the first layer and the inputted channel of the second layer of the hardVFE is set as 64 in the configuration. I consider the code in mmdet3d/models/voxel_encoders/voxel_encoder.py#L357 unnecessary. Could you provide some suggestions or explanations? Thanks

SeaBird-1995, 2023-05-29:
@FrontierBreaker I also encountered this problem, have you solved this error?

FrontierBreaker, 2023-05-29:
Not yet.