ADLab-AutoDrive/BEVFusion #59: traing fusion stream warning: missing keys in source state_dict

zjufkq, 2023-03-06:
I have the same problem as the issue [https://github.com/ADLab-AutoDrive/BEVFusion/issues/23](url)
I use the weight parameter provided by the author cam_pp.pth [https://drive.google.com/file/d/1OeDjGoKpU1FQ42_18pJZ-UqwxLDc6HRc/view](url) and hv_ pointpilars...pth [https://drive.google.com/file/d/1CBF0g1i23hfS-KRihX9J6QF2CglbWjsM/view](url) (github) ，then runs the fusion stream code (run this command)
./tools/dist_ train.sh configs/bevfusion/bevf_ pp_ 2x8_ 1x_ nusc.py 8
The above warning will appear:
mmdet - WARNING - The model and loaded state dict do not match exactly
**missing keys in source state_dict:** img_backbone.cb_modules.0.patch_embed.proj.weight, img_backbone.cb_modules.0.patch_embed.proj.bias, img_backbone.cb_modules.0.patch_embed.norm.weight, img_backbone.cb_modules.0.patch_embed.norm.bias, img_backbone.cb_modules.0.layers.0.blocks.0.norm1.weight, img_backbone.cb_modules.0.layers.0.blocks.0.norm1.bias, img_backbone.cb_modules.0.layers.0.blocks.0.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.0.blocks.0.attn.relative_position_index, img_backbone.cb_modules.0.layers.0.blocks.0.attn.qkv.weight, img_backbone.cb_modules.0.layers.0.blocks.0.attn.qkv.bias, img_backbone.cb_modules.0.layers.0.blocks.0.attn.proj.weight, img_backbone.cb_modules.0.layers.0.blocks.0.attn.proj.bias, img_backbone.cb_modules.0.layers.0.blocks.0.norm2.weight, img_backbone.cb_modules.0.layers.0.blocks.0.norm2.bias, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc1.weight, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc1.bias, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc2.weight, img_backbone.cb_modules.0.layers.0.blocks.0.mlp.fc2.bias, img_backbone.cb_modules.0.layers.0.blocks.1.norm1.weight, img_backbone.cb_modules.0.layers.0.blocks.1.norm1.bias, img_backbone.cb_modules.0.layers.0.blocks.1.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.0.blocks.1.attn.relative_position_index, img_backbone.cb_modules.0.layers.0.blocks.1.attn.qkv.weight, img_backbone.cb_modules.0.layers.0.blocks.1.attn.qkv.bias, img_backbone.cb_modules.0.layers.0.blocks.1.attn.proj.weight, img_backbone.cb_modules.0.layers.0.blocks.1.attn.proj.bias, img_backbone.cb_modules.0.layers.0.blocks.1.norm2.weight, img_backbone.cb_modules.0.layers.0.blocks.1.norm2.bias, img_backbone.cb_modules.0.layers.0.blocks.1.mlp.fc1.weight, img_backbone.cb_modules.0.layers.0.blocks.1.mlp.fc1.bias, img_backbone.cb_modules.0.layers.0.blocks.1.mlp.fc2.weight, img_backbone.cb_modules.0.layers.0.blocks.1.mlp.fc2.bias, img_backbone.cb_modules.0.layers.0.downsample.reduction.weight, img_backbone.cb_modules.0.layers.0.downsample.norm.weight, img_backbone.cb_modules.0.layers.0.downsample.norm.bias, img_backbone.cb_modules.0.layers.1.blocks.0.norm1.weight, img_backbone.cb_modules.0.layers.1.blocks.0.norm1.bias, img_backbone.cb_modules.0.layers.1.blocks.0.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.1.blocks.0.attn.relative_position_index, img_backbone.cb_modules.0.layers.1.blocks.0.attn.qkv.weight, img_backbone.cb_modules.0.layers.1.blocks.0.attn.qkv.bias, img_backbone.cb_modules.0.layers.1.blocks.0.attn.proj.weight, img_backbone.cb_modules.0.layers.1.blocks.0.attn.proj.bias, img_backbone.cb_modules.0.layers.1.blocks.0.norm2.weight, img_backbone.cb_modules.0.layers.1.blocks.0.norm2.bias, img_backbone.cb_modules.0.layers.1.blocks.0.mlp.fc1.weight, img_backbone.cb_modules.0.layers.1.blocks.0.mlp.fc1.bias, img_backbone.cb_modules.0.layers.1.blocks.0.mlp.fc2.weight, img_backbone.cb_modules.0.layers.1.blocks.0.mlp.fc2.bias, img_backbone.cb_modules.0.layers.1.blocks.1.norm1.weight, img_backbone.cb_modules.0.layers.1.blocks.1.norm1.bias, img_backbone.cb_modules.0.layers.1.blocks.1.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.1.blocks.1.attn.relative_position_index, img_backbone.cb_modules.0.layers.1.blocks.1.attn.qkv.weight, img_backbone.cb_modules.0.layers.1.blocks.1.attn.qkv.bias, img_backbone.cb_modules.0.layers.1.blocks.1.attn.proj.weight, img_backbone.cb_modules.0.layers.1.blocks.1.attn.proj.bias, img_backbone.cb_modules.0.layers.1.blocks.1.norm2.weight, img_backbone.cb_modules.0.layers.1.blocks.1.norm2.bias, img_backbone.cb_modules.0.layers.1.blocks.1.mlp.fc1.weight, img_backbone.cb_modules.0.layers.1.blocks.1.mlp.fc1.bias, img_backbone.cb_modules.0.layers.1.blocks.1.mlp.fc2.weight, img_backbone.cb_modules.0.layers.1.blocks.1.mlp.fc2.bias, img_backbone.cb_modules.0.layers.1.downsample.reduction.weight, img_backbone.cb_modules.0.layers.1.downsample.norm.weight, img_backbone.cb_modules.0.layers.1.downsample.norm.bias, img_backbone.cb_modules.0.layers.2.blocks.0.norm1.weight, img_backbone.cb_modules.0.layers.2.blocks.0.norm1.bias, img_backbone.cb_modules.0.layers.2.blocks.0.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.2.blocks.0.attn.relative_position_index, img_backbone.cb_modules.0.layers.2.blocks.0.attn.qkv.weight, img_backbone.cb_modules.0.layers.2.blocks.0.attn.qkv.bias, img_backbone.cb_modules.0.layers.2.blocks.0.attn.proj.weight, img_backbone.cb_modules.0.layers.2.blocks.0.attn.proj.bias, img_backbone.cb_modules.0.layers.2.blocks.0.norm2.weight, img_backbone.cb_modules.0.layers.2.blocks.0.norm2.bias, img_backbone.cb_modules.0.layers.2.blocks.0.mlp.fc1.weight, img_backbone.cb_modules.0.layers.2.blocks.0.mlp.fc1.bias, img_backbone.cb_modules.0.layers.2.blocks.0.mlp.fc2.weight, img_backbone.cb_modules.0.layers.2.blocks.0.mlp.fc2.bias, img_backbone.cb_modules.0.layers.2.blocks.1.norm1.weight, img_backbone.cb_modules.0.layers.2.blocks.1.norm1.bias, img_backbone.cb_modules.0.layers.2.blocks.1.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.2.blocks.1.attn.relative_position_index, img_backbone.cb_modules.0.layers.2.blocks.1.attn.qkv.weight, img_backbone.cb_modules.0.layers.2.blocks.1.attn.qkv.bias, img_backbone.cb_modules.0.layers.2.blocks.1.attn.proj.weight, img_backbone.cb_modules.0.layers.2.blocks.1.attn.proj.bias, img_backbone.cb_modules.0.layers.2.blocks.1.norm2.weight, img_backbone.cb_modules.0.layers.2.blocks.1.norm2.bias, img_backbone.cb_modules.0.layers.2.blocks.1.mlp.fc1.weight, img_backbone.cb_modules.0.layers.2.blocks.1.mlp.fc1.bias, img_backbone.cb_modules.0.layers.2.blocks.1.mlp.fc2.weight, img_backbone.cb_modules.0.layers.2.blocks.1.mlp.fc2.bias, img_backbone.cb_modules.0.layers.2.blocks.2.norm1.weight, img_backbone.cb_modules.0.layers.2.blocks.2.norm1.bias, img_backbone.cb_modules.0.layers.2.blocks.2.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.2.blocks.2.attn.relative_position_index, img_backbone.cb_modules.0.layers.2.blocks.2.attn.qkv.weight, img_backbone.cb_modules.0.layers.2.blocks.2.attn.qkv.bias, img_backbone.cb_modules.0.layers.2.blocks.2.attn.proj.weight, img_backbone.cb_modules.0.layers.2.blocks.2.attn.proj.bias, img_backbone.cb_modules.0.layers.2.blocks.2.norm2.weight, img_backbone.cb_modules.0.layers.2.blocks.2.norm2.bias, img_backbone.cb_modules.0.layers.2.blocks.2.mlp.fc1.weight, img_backbone.cb_modules.0.layers.2.blocks.2.mlp.fc1.bias, img_backbone.cb_modules.0.layers.2.blocks.2.mlp.fc2.weight, img_backbone.cb_modules.0.layers.2.blocks.2.mlp.fc2.bias, img_backbone.cb_modules.0.layers.2.blocks.3.norm1.weight, img_backbone.cb_modules.0.layers.2.blocks.3.norm1.bias, img_backbone.cb_modules.0.layers.2.blocks.3.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.2.blocks.3.attn.relative_position_index, img_backbone.cb_modules.0.layers.2.blocks.3.attn.qkv.weight, img_backbone.cb_modules.0.layers.2.blocks.3.attn.qkv.bias, img_backbone.cb_modules.0.layers.2.blocks.3.attn.proj.weight, img_backbone.cb_modules.0.layers.2.blocks.3.attn.proj.bias, img_backbone.cb_modules.0.layers.2.blocks.3.norm2.weight, img_backbone.cb_modules.0.layers.2.blocks.3.norm2.bias, img_backbone.cb_modules.0.layers.2.blocks.3.mlp.fc1.weight, img_backbone.cb_modules.0.layers.2.blocks.3.mlp.fc1.bias, img_backbone.cb_modules.0.layers.2.blocks.3.mlp.fc2.weight, img_backbone.cb_modules.0.layers.2.blocks.3.mlp.fc2.bias, img_backbone.cb_modules.0.layers.2.blocks.4.norm1.weight, img_backbone.cb_modules.0.layers.2.blocks.4.norm1.bias, img_backbone.cb_modules.0.layers.2.blocks.4.attn.relative_position_bias_table, img_backbone.cb_modules.0.layers.2.blocks.4.attn.relative_position_index, img_backbone.cb_modules.0.layers.2.blocks.4.attn.qkv.weight, img_backbone.cb_modules.0.layers.2.blocks.4.attn.qkv.bias, img_backbone.cb_modules.0.layers.2.blocks.4.attn.proj.weight, img_ [... cut ...] t_splat_shot_vis.bevencode.7.bias, lift_splat_shot_vis.bevencode.7.running_mean, lift_splat_shot_vis.bevencode.7.running_var, lift_splat_shot_vis.bevencode.9.weight, lift_splat_shot_vis.bevencode.10.weight, lift_splat_shot_vis.bevencode.10.bias, lift_splat_shot_vis.bevencode.10.running_mean, lift_splat_shot_vis.bevencode.10.running_var, seblock.att.1.weight, seblock.att.1.bias, reduc_conv.conv.weight, reduc_conv.bn.weight, reduc_conv.bn.bias, reduc_conv.bn.running_mean, reduc_conv.bn.running_var

Crayon-new, 2023-05-02:
Hi, have you solved it?

maxiuw, 2024-04-22:
Correct me if I am wrong but I think what is happening is
1. You load the cam backbone.
2. You are loading the whole model (which actually is just a LiDAR backbone) with strict=False, thus a lot of keys/weights are missing but you already loaded the camera weights before (use debugger and check, did you get any errors at that stage?).