ADLab-AutoDrive/BEVFusion #83: Excuse me.There is a problem with the preprocessing mini's nuscence dataset, how should I solve it？

lymcs, 2023-09-20:
:~/bevfusion$ python tools/create_data.py  nuscenes --root-path ./data/nuscenes/ --version v1.0-mini --out-dir data/nuscenes/ --extra-tag nuscenes
======
Loading NuScenes tables for version v1.0-mini...
23 category,
8 attribute,
4 visibility,
911 instance,
12 sensor,
120 calibrated_sensor,
31206 ego_pose,
8 log,
10 scene,
404 sample,
31206 sample_data,
18538 sample_annotation,
4 map,
Done loading in 0.266 seconds.
======
Reverse indexing ...
Done reverse indexing in 0.0 seconds.
======
total scene num: 10
exist scene num: 10
train scene: 8, val scene: 2
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>] 404/404, 3.4 task/s, elapsed: 119s, ETA:     0s
nuscenes
train sample: 323, val sample: 81
Create GT Database of NuScenesDataset
Traceback (most recent call last):
  File "/home/wsd2/anaconda3/envs/bevfusion_mit/lib/python3.8/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg
    return obj_cls(**args)
  File "/home/wsd2/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 143, in __init__
    super().__init__(
  File "/home/wsd2/bevfusion/mmdet3d/datasets/custom_3d.py", line 66, in __init__
    self.data_infos = self.load_annotations(self.ann_file)
  File "/home/wsd2/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 202, in load_annotations
    data = mmcv.load(ann_file)
  File "/home/wsd2/anaconda3/envs/bevfusion_mit/lib/python3.8/site-packages/mmcv/fileio/io.py", line 60, in load
    with BytesIO(file_client.get(file)) as f:
  File "/home/wsd2/anaconda3/envs/bevfusion_mit/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 992, in get
    return self.client.get(filepath)
  File "/home/wsd2/anaconda3/envs/bevfusion_mit/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 517, in get
    with open(filepath, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'data/nuscenes//nuscenes_infos_train.pkl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/create_data.py", line 122, in <module>
    nuscenes_data_prep(
  File "tools/create_data.py", line 45, in nuscenes_data_prep
    create_groundtruth_database(
  File "/home/wsd2/bevfusion/tools/data_converter/create_gt_database.py", line 252, in create_groundtruth_database
    dataset = build_dataset(dataset_cfg)
  File "/home/wsd2/bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset
    dataset = build_from_cfg(cfg, DATASETS, default_args)
  File "/home/wsd2/anaconda3/envs/bevfusion_mit/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
FileNotFoundError: NuScenesDataset: [Errno 2] No such file or directory: 'data/nuscenes//nuscenes_infos_train.pkl'

Wangquans, 2024-02-19:
Did you solve it? I have the same problem:
FileNotFoundError: [Errno 2] No such file or directory: 'data/nuscenes//nuscenes_infos_train.pkl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/create_data.py", line 122, in <module>
    nuscenes_data_prep(
  File "tools/create_data.py", line 45, in nuscenes_data_prep
    create_groundtruth_database(
  File "/media/robot/DB87DB580E5CF130/MIT-bevfusion/tools/data_converter/create_gt_database.py", line 252, in create_groundtruth_database
    dataset = build_dataset(dataset_cfg)
  File "/media/robot/DB87DB580E5CF130/MIT-bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset
    dataset = build_from_cfg(cfg, DATASETS, default_args)
  File "/home/robot/software/anaconda3/envs/mit-bevfusion/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
FileNotFoundError: NuScenesDataset: [Errno 2] No such file or directory: 'data/nuscenes//nuscenes_infos_train.pkl'

Barath19, 2024-07-04:
> Did you solve it? I have the same problem: FileNotFoundError: [Errno 2] No such file or directory: 'data/nuscenes//nuscenes_infos_train.pkl'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "tools/create_data.py", line 122, in nuscenes_data_prep( File "tools/create_data.py", line 45, in nuscenes_data_prep create_groundtruth_database( File "/media/robot/DB87DB580E5CF130/MIT-bevfusion/tools/data_converter/create_gt_database.py", line 252, in create_groundtruth_database dataset = build_dataset(dataset_cfg) File "/media/robot/DB87DB580E5CF130/MIT-bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset dataset = build_from_cfg(cfg, DATASETS, default_args) File "/home/robot/software/anaconda3/envs/mit-bevfusion/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg raise type(e)(f'{obj_cls.**name**}: {e}') FileNotFoundError: NuScenesDataset: [Errno 2] No such file or directory: 'data/nuscenes//nuscenes_infos_train.pkl'

remove / from the data_root