ADLab-AutoDrive/BEVFusion #86: 找不到'work_dirs/mask_rcnn_dbswin-t_fpn_3x_nuim_cocopre/epoch_36.pth'

chenduo111, 2023-11-18:
你好，我在训练时显示找不到是这个FileNotFoundError: [Errno 2] No such file or directory: 'work_dirs/mask_rcnn_dbswin-t_fpn_3x_nuim_cocopre/epoch_36.pth'
请问这个文件需要在哪里下载

eatchips, 2024-03-24:
解决了么哥？我也没有这个文件

Libraaer, 2024-08-31:
> 你好，我在训练时显示找不到是这个FileNotFoundError: [Errno 2] No such file or directory: 'work_dirs/mask_rcnn_dbswin-t_fpn_3x_nuim_cocopre/epoch_36.pth' 请问这个文件需要在哪里下载

请问找到了吗，我也需要这个文件

Libraaer, 2024-08-31:
> 解决了么哥？我也没有这个文件

请问找到了吗，我也需要这个文件

eatchips, 2024-08-31:
抱歉，后续我也没找到解决办法，也没有继续做这个方向

---- 回复的原邮件 ----
| 发件人 | Yan ***@***.***> |
| 日期 | 2024年08月31日 16:18 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [ADLab-AutoDrive/BEVFusion] 找不到'work_dirs/mask_rcnn_dbswin-t_fpn_3x_nuim_cocopre/epoch_36.pth' (Issue #86) |

你好，我在训练时显示找不到是这个FileNotFoundError: [Errno 2] No such file or directory: 'work_dirs/mask_rcnn_dbswin-t_fpn_3x_nuim_cocopre/epoch_36.pth' 请问这个文件需要在哪里下载

请问找到了吗，我也需要这个文件

2500490261, 2024-12-12:
替换成/home/tcj24/code/3D_Object_Detection/BEVFusion-bv2/pretrained/swint-nuimages-pretrained.pth