ADLab-AutoDrive/BEVFusion #91: DockerFile有问题

OrangeJuice3211, 2024-03-12:
位于docker文件夹下的DockerFile，使用docker build . -t bevfusion创建docker镜像时，出现了一堆问题，详细问题如下：
1. 运行到32行docker build . -t bevfusion时，报错
[image]
2. 注释掉这一行，在此构建镜像，又报错：
~~~python
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements/build.txt'
~~~
3. 发现这个TXT里面啥也没有，遂又注释掉，在此构建，又报错：
~~~python
ERROR: File "setup.py" not found. Directory cannot be installed in editable mode: /workspace
~~~
所以这个DockerFile是能用的吗？？？

OrangeJuice3211, 2024-03-13:
修改了一堆问题之后，出现了最终错误，能到DockerFile第42行`RUN pip install mmpycocotools --no-cache-dir --force --no-deps
`，麻了，放弃了
[image]
[image]

Barath19, 2024-06-27:
RUN pip install cython==0.29.33
RUN pip install mmpycocotools --no-cache-dir --force --no-deps

cheopis, 2024-09-10:
I don't know if this wold still be useful, but I managed to fix some of your problems:
1.  > RUN pip install nuscenes-devkit einops returning:
ERROR: Could not find a version that satisfies the requirement numpy>=1.22.0

    Change the line `RUN pip install nuscenes-devkit einops` in Dockerfile to 
```python
RUN pip install nuscenes-devkit==1.1.9 einops
``` 

3.  >ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements/build.txt'
ERROR: File "setup.py" not found. Directory cannot be installed in editable mode: /workspace

   Comment `RUN pip install -r ../requirements/build.txt` and change `RUN pip install --no-cache-dir -e .` to:
```python
RUN git clone https://github.com/ADLab-AutoDrive/BEVFusion /BEVFusion \
    && cd /BEVFusion \
    && pip install -r requirements/build.txt \
    && pip install --no-cache-dir -e .
``` 

The `pip install -r requirements/build.txt` indeed looks like it's done nothing, but I decided to keep it in the code because why not.

4. >RUN pip install mmpycocotools --no-cache-dir --force --no-deps

Resolved by changing `RUN pip install cython` as described by @Barath19 

```python
RUN pip install cython==0.29.33
``` 

5. I also got an error with line `RUN find . -name "*.so" | xargs rm`, but just commented it